RobotWebTools/rosbridge_suite #920: cancel_action_goal results in action_result with no status

mirons5678, 2024-04-12:
**Description**

If I send `send_action_goal` message and let the action run to completion I get back an `action_result` with a `values` field that contains a `status` of 4 (i.e. SUCCEEDED), e.g.:

```json
{
  "op": "action_result",
  "action": "/robot/navigate_to_pose",
  "values": {
    "status": 4,
    "result": {
      "result": {}
    }
  },
  "result": true,
  "id": "2"
}
```

However, if I send a `cancel_action_goal` message before the action has completed, I get back an `action_result` with a `values` field that does not contain a `status` (I expected to see `status` 5 (i.e. CANCELED)).

- Library Version: Built from source (https://github.com/RobotWebTools/rosbridge_suite/commit/7d78af16d30d0ffe232abcc65d0928ce90bd61f7)
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**

Send `send_action_goal` followed by `cancel_action_goal`. Am running rosbridge with `send_action_goals_in_new_thread:=true`

**Expected Behavior**

Expect to receive an `action_result` with `values` field with `status` 5 (CANCELED), e.g.

```json
{
  "op": "action_result",
  "action": "/robot/navigate_to_pose",
  "values": {
    "status": 5,
    "result": {
      "result": {}
    }
  },
  "result": true,
  "id": "2"
}
```

**Actual Behavior**

Actually receive an `action_result` with `values` field with no `status`, e.g.

```json
{
  "op": "action_result",
  "action": "/robot/navigate_to_pose",
  "values": "None",
  "result": false,
  "id": "1"
}
```

Also see the following error log from the rosbridge server:

```
[rosbridge_websocket-1] [ERROR] [1712917839.725245716] [rosbridge_websocket]: [Client c19854d4-a8e8-49d2-9736-3b500c846d61] send_action_goal Exception: send_action_goal:/robot/navigate_to_pose:49724a1e-dffc-4a4d-a8f6-a66cbe77d5ec
```

sea-bass, 2024-09-28:
I tried something to fix this, but I couldn't get it working with CANCELED. If you're ok with it returning ABORTED and no result, there is https://github.com/RobotWebTools/rosbridge_suite/pull/952

sea-bass, 2024-09-28:
I think the issue with my above attempt, which didn't quite work, is that if you post an `action_result` message, it seems to always succeed the goal.

The current workaround of aborting a goal when a cancel is sent doesn't seem correct; rather, we should wait until the action returns, check its status, and update the goal in rosbridge accordingly. I may have another idea to try when I can find some more time.

sea-bass, 2024-09-29:
Update: I just tried again in https://github.com/RobotWebTools/rosbridge_suite/pull/953 -- could you check if this works for you?

PR #953 (2024-10-02): Fix action cancellation by passing status from JSON
**Public API Changes**
This slightly changes the protocol, specifically that the `action_result` message now contains an additional `status` field which contains an integer corresponding to the enum in https://docs.ros2.org/latest/api/action_msgs/msg/GoalStatus.html.

This does mean that the client libraries (namely, `roslibjs` which has the only working implementation) need to be updated as well.

**Description**
As described above, this PR adds a new `status` field to the `action_result` message to handle action cancellation properly.

---
 [... cut]
closed 2024-10-02 by sea-bass