RobotWebTools/rosbridge_suite #921: Not working on Windows 11 with Foxy

Saad-Naseer, 2024-04-18:
**Description**
There is no error in building process of rosbridge_suite but when i run the launch file it is looking for .exe file. I am using the command colcon build --merge-install i will create the lib folder and inside this lib folder i have .exe files for all of my packages but it will not generate the .exe file for rosbridge_server.

- Library Version: 1.3.1
- ROS Version: Foxy
- Platform / OS: Windows 11

**Steps To Reproduce**

Finished <<< siam_mask [23.2s]
Finished <<< rosbridge_library [29.8s]
Starting >>> rosapi
Finished <<< compressed_depth_image_transport [41.3s]
Finished <<< compressed_image_transport [40.8s]
Starting >>> image_transport_plugins
Finished <<< rosapi [13.1s]
Starting >>> rosbridge_server
Finished <<< image_pub_sub [49.7s]
Starting >>> tracking_gui
Finished <<< image_transport_plugins [7.41s]
Finished <<< rosbridge_server [9.69s]
Starting >>> rosbridge_suite
Finished <<< rosbridge_suite [2.97s]
Finished <<< tracking_gui [36.3s]
Starting >>> app_launch
Finished <<< app_launch [6.59s]

Summary: 34 packages finished [3min 39s]

**Expected Behavior**
.exe should be generated for rosbridge_websocket in Lib folder.

**Actual Behavior**
it only generates rosbridge_websocket.py in and rosbridge_websocket.(no extension)

sea-bass, 2024-05-08:
ROS 2 Foxy has been unsupported since June 2023, and also was only supported on Windows 10, not 11: https://docs.ros.org/en/foxy/Releases/Release-Foxy-Fitzroy.html#supported-platforms

In fact, seems like even the latest version of ROS 2 still only has Windows 10 as a Tier 1 platform.

You may be able to roll back to an older version of this repo, but I think there are no guarantees with your particular setup, sorry to say.

EricGallimore, 2024-07-16:
Take a look at the discussion in #914, which is functionally a duplicate of this.

tldr; several `rosbridge` packages need to be converted to use `ament_python` instead of `ament_cmake` to work on Windows (since `ament_python` can generate `exe` wrappers, but `ament_cmake` does not).