RobotWebTools/rosbridge_suite #926: get_parameter_value not present in Jazzy for rosapi_node

MoffKalast, 2024-05-09:
**Description**
Apparently the API has changed in the new release:

```bash
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/rosapi/rosapi_node", line 41, in <module>
    from rosapi import glob_helper, objectutils, params, proxy
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rosapi/params.py", line 41, in <module>
    from ros2param.api import call_get_parameters, call_set_parameters, get_parameter_value
ImportError: cannot import name 'get_parameter_value' from 'ros2param.api' (/opt/ros/jazzy/lib/python3.12/site-packages/ros2param/api/__init__.py)
[ros2run]: Process exited with failure 1

```

- Library Version: ros-jazzy-rosbridge-suite:arm64/noble 1.3.2-3noble.20240502.173308 uptodate
- ROS Version: Jazzy
- Platform / OS: Ubuntu 24.04 Server, aarch64

**Steps To Reproduce**
`ros2 run rosapi rosapi_node`

**Expected Behavior**
Rosapi doesn't crash.

**Actual Behavior**
Rosapi crashes. :smiley:

earwole1, 2024-05-30:
@MoffKalast 

We are attempting to upgrade from humble to jazzy today and this also breaks `rosbridge_server`, as in:
`ros2 launch rosbridge_server rosbridge_websocket_launch.xml` 

Looks like this change was made back in Nov 2023: see [here](https://github.com/ros2/ros2cli/pull/866)

The solution appears to be to import from rclpy:
`from  rclpy.parameter import get_parameter_value`

Tested this locally and the behavior has been resolved. Just need to add a PR

rubenanapu, 2024-06-19:
> The solution appears to be to import from rclpy:
>  `from rclpy.parameter import get_parameter_value`

Awesome.

The file I added the suggested import was:

`/opt/ros/jazzy/lib/python3.12/site-packages/ros2param/api/__init__.py`

It worked like a charm.

sea-bass, 2024-07-14:
Closed by #932 -- thanks again @Oberacda!

closed 2024-07-14 by sea-bass