RobotWebTools/rosbridge_suite #930: Subscriber to topic throughput decrease eventually

oriolorra, 2024-05-31:
Hi,

I have encountered some odd behavior. At some point, a subscriber topic decrease throughput, when the publishing topic does not decrease the throughput.

1. I have a cpp node publishing a string representing a base64 image at 10Hz, which its rate does not decrease over time. Checked via **ros2 topic hz /b64_image**.
2. Rosbridge_server is launched too, via **ros2 launch rosbridge_server rosbridge_websocket_launch.xml**.
3. A webpage is receiving **/b64_image** with roslibjs, and the image is rendered into a canvas. Eventually, the frame rate (fps) of message received decrease almost to 0 fps (image attached). If, I refresh manually, the webpage returns to 10fps but return to 0 fps a few seconds later. 

Roslibjs topic is configured like this:

```
 var listener = new ROSLIB.Topic({
    ros : ros,
    name : '/b64_image',
    messageType : 'std_msgs/String',
    queue_size: 1,
    throttle_rate: 10
    })
```
[image]

- Library Version: 1.3.2-1jammy.20240517.175148
- ROS Version: ROS2 Humble
- Platform / OS: Ubuntu 22.04

**Expected Behavior**
I expect to get an almost constant frame rate at 10 fps at the webpage, let's say forever.

**Actual Behavior**
Webpage has 10 fps during only a few seconds/minuts.

Thanks

flxinxout, 2025-06-04:
same issue. I have a camera oakd-lite on my robot. it communicates via wifi and I get around 1Hz at some point on my client side, while 10Hz is still good on my robot side. I use compressed IMage

flxinxout, 2025-06-07:
@joshjowen thanks for the reply. but do you have any idea why this happens?