RobotWebTools/rosbridge_suite #931: ROS 2 send_action_goal Operation not Recognized

TheGuyWithoutH, 2024-06-29:
**Description**
After installing rosbridge_server with apt-get, I try to run the example of [action client in ROS 2](https://github.com/RobotWebTools/roslibjs/blob/develop/examples/ros2_action_client.html) from roslibjs. The ros client is connected but then when the example code tries to send the action request, the following message appears on the rosbridge-server terminal:

```(bash)
[rosbridge_websocket-1] [INFO] [1719650740.191282598] [rosbridge_websocket]: Calling services in existing thread
[rosbridge_websocket-1] [INFO] [1719650740.192217869] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [ERROR] [1719650740.192975287] [rosbridge_websocket]: [Client b6ab93b5-4dd4-414e-9655-a7eb2581237d] [id: send_action_goal:/fibonacci:1] Unknown operation: send_action_goal.  Allowed operations: ['call_service', 'advertise', 'unadvertise', 'publish', 'subscribe', 'unsubscribe', 'fragment', 'advertise_service', 'service_response', 'unadvertise_service']
```

Given the list of allowed operations, it seems that the actions of the rosbridge protocol are not included. How can it be? I checked the Protocol documentation that explicitly talks about `send_action_goal`, so I am confused. 

- Library Version: 1.3.2-1jammy.20240530.094238
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04.4 LTS

**Steps To Reproduce**
- `sudo apt-get install ros-humble-rosbridge-server`
- `ros2 launch rosbridge_server rosbridge_websocket_launch.xml`

- `git clone git@github.com:RobotWebTools/roslibjs.git`
- `npm i`

- ros2 run action_tutorials_py fibonacci_action_server
- open the HTML page at roslibjs/examples/ros2_action_client.html

vcharpenay, 2024-07-01:
I ran into the same problem but it seems the current release simply doesn't include actions. PR #886 was merged after the the code base was tagged `v1.3.2` (Nov 10th last year vs. Sept 7th).

TheGuyWithoutH, 2024-07-01:
Okay thanks! I will try installing directly from the repo. It seemed weird that this feature is already mentionned in other documentation without being released yet.

Is there a release date planned?

sea-bass, 2024-08-16:
For a bit of background, this project's action supports were driven by needing them in `roslibjs`.

The working plan is that we need some help writing integration tests for ROS 2 in `roslibjs` (https://github.com/RobotWebTools/roslibjs/issues/746), and once that is clear, we were going to release a version 2 of both these repos.

closed 2024-08-16 by TheGuyWithoutH

sea-bass, 2024-12-07:
FYI `rosbridge` 2.0.0 should now be available from binaries in Humble!