RobotWebTools/rosbridge_suite #990: After running for a period of time, a certain topic may not be subscribed to

nvidia-tzkj, 2025-01-06:
Recently, when using rosbridge_server, I found that occasionally there may be situations where a certain topic is not subscribed. We use several different clients to establish connections with it. Other status subscriptions are normal, but robot_pose displays subscribed but does not return data. This topic currently has the highest frequency among all topics, at 20Hz

- Library Version: 
- ROS Version:  ROS1
- Platform / OS: ubuntu 20.04

this is our param setting:
<launch>
  <arg name="port" default="9090" />
  <arg name="address" default="0.0.0.0" />
  <arg name="ssl" default="false" />
  <arg name="certfile" default=""/>
  <arg name="keyfile" default="" />

  <arg name="retry_startup_delay" default="5" />

  <arg name="fragment_timeout" default="600" />
  <arg name="delay_between_messages" default="0" />
  <arg name="max_message_size" default="None" />
  <arg name="unregister_timeout" default="10000000" />
  <arg name="websocket_external_port" default="None" />

  <arg name="use_compression" default="false" />

  <arg name="authenticate" default="false" />

  <arg name="websocket_ping_interval" default="0" />
  <arg name="websocket_ping_timeout" default="30" />
  <arg name="websocket_null_origin" default="true" />

  <arg name="topics_glob" default="[*]" />
  <arg name="services_glob" default="[*]" />
  <arg name="params_glob" default="[*]" />
  <arg name="bson_only_mode" default="false" />
  
  <arg name="output" default="screen" />

  
  <arg unless="$(arg bson_only_mode)" name="binary_encoder" default="default"/>

  <group if="$(arg ssl)">
    <node name="rosbridge_websocket" pkg="rosbridge_server" type="rosbridge_websocket" output="$(arg output)">
      <param name="certfile" value="$(arg certfile)" />
      <param name="keyfile" value="$(arg keyfile)" />
      <param name="authenticate" value="$(arg authenticate)" />
      <param name="port" value="$(arg port)"/>
      <param name="address" value="$(arg address)"/>
      <param name="retry_startup_delay" value="$(arg retry_startup_delay)"/>
      <param name="fragment_timeout" value="$(arg fragment_timeout)"/>
      <param name="delay_between_messages" value="$(arg delay_between_messages)"/>
      <param name="max_message_size" value="$(arg max_message_size)"/>
      <param name="unregister_timeout" value="$(arg unregister_timeout)"/>
      <param name="use_compression" value="$(arg use_compression)"/>

      <param name="websocket_ping_interval" value="$(arg websocket_ping_interval)" />
      <param name="websocket_ping_timeout" value="$(arg websocket_ping_timeout)" />
      <param name="websocket_external_port" value="$(arg websocket_external_port)" />
      <param name="websocket_null_origin" value="$(arg websocket_null_origin)" />

      <param name="topics_glob" value="$(arg topics_glob)"/>
      <param name="services_glob" value="$(arg services_glob)"/>
      <param name="params_glob" value="$(arg params_glob)"/>
    </node>
  </group>
  <group unless="$(arg ssl)">
    <node name="rosbridge_websocket" pkg="rosbridge_server" type="rosbridge_websocket" output="$(arg output)">
      <param name="authenticate" value="$(arg authenticate)" />
      <param name="port" value="$(arg port)"/>
      <param name="address" value="$(arg address)"/>
      <param name="retry_startup_delay" value="$(arg retry_startup_delay)"/>
      <param name="fragment_timeout" value="$(arg fragment_timeout)"/>
      <param name="delay_between_messages" value="$(arg delay_between_messages)"/>
      <param name="max_message_size" value="$(arg max_message_size)"/>
      <param name="unregister_timeout" value="$(arg unregister_timeout)"/>
      <param name="use_compression" value="$(arg use_compression)"/>

      <param name="websocket_ping_interval" value="$(arg websocket_ping_interval)" />
      <param name="websocket_ping_timeout" value="$(arg websocket_ping_timeout)" />
      <param name="websocket_external_port" value="$(arg websocket_external_port)" />

      <param name="topics_glob" value="$(arg topics_glob)"/>
      <param name="services_glob" value="$(arg services_glob)"/>
      <param name="params_glob" value="$(arg params_glob)"/>

      <param name="bson_only_mode" value="$(arg bson_only_mode)"/>
    </node>
  </group>

  <node name="rosapi" pkg="rosapi" type="rosapi_node" output="$(arg output)">
    <param name="topics_glob" value="$(arg topics_glob)"/>
    <param name="services_glob" value="$(arg services_glob)"/>
    <param name="params_glob" value="$(arg params_glob)"/>
  </node>
</launch>

nvidia-tzkj, 2025-01-06:
At the beginning of starting rosbridge, all subscriptions were normal, but after running for a period of time, the robot_pose topic no longer pushes. Even if a connection is re established through rosbridge, the topic information from the device cannot be received anymore, but other topic information from the device can be received.

bjsowa, 2025-02-25:
@nvidia-tzkj Could you check if #1004 fixed this issue for you?