RobotWebTools/rosbridge_suite #1022: Offering incompatible QoS

Egoless41, 2025-03-27:
**Description**
When a topic has either a transient_local durability policy or a reliable reliability policy, rosbridge creates a subscriber with the same policies. This causes the client to be unable to subscribe to the topic.

```
[rosbridge_websocket]: New publisher discovered on topic '/tf', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY
```   

- Library Version: 2.2.0
- ROS Version: Jazzy
- Platform / OS: Ubuntu 24.04 inside docker container

**Steps To Reproduce**

When launching rosbridge with the Controller Manager library, the /tf topic encounters this issue.

**Expected Behavior**
Even if a publisher has a strict policy, the subscriber should still be allowed to use a less strict one.

**Actual Behavior**
Subscriber with strict policy refuses to subscribe less strict publishers.

bjsowa, 2025-03-27:
I'm not sure I follow. Am I currently assuming that in your situation rosbridge is subscribing with Reliable policy when there are publishers with Best effort policy? This should not happen after #991 was merged as far as I understand.

Could you show me the output of:
```
ros2 topic info -v /tf
```

Egoless41, 2025-03-28:
Actually my issue is not about Reliability policy, my problem is Durability policy. Here is my output. 

```
ros2 topic info /tf --v
Type: tf2_msgs/msg/TFMessage

Publisher count: 5

Node name: ekf_se_odom
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: PUBLISHER
GID: 01.0f.fb.07.53.00.91.ca.00.00.00.00.00.00.1b.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: ekf_se_map
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: PUBLISHER
GID: 01.0f.fb.07.54.00.28.df.00.00.00.00.00.00.1b.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: base_link_appender
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: PUBLISHER
GID: 01.0f.fb.07.5d.00.e4.00.00.00.00.00.00.00.14.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: robot_state_publisher
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: PUBLISHER
GID: 01.0f.fb.07.b5.00.aa.1f.00.00.00.00.00.00.14.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: velocity_controller
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: PUBLISHER
GID: 01.0f.fb.07.b7.00.78.88.00.00.00.00.00.00.a6.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 5

Node name: transform_listener_impl_5586ae756ec0
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: SUBSCRIPTION
GID: 01.0f.fb.07.53.00.91.ca.00.00.00.00.00.00.18.04
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: transform_listener_impl_557b20fd2f00
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: SUBSCRIPTION
GID: 01.0f.fb.07.54.00.28.df.00.00.00.00.00.00.18.04
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: transform_listener_impl_55858b0bfd40
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: SUBSCRIPTION
GID: 01.0f.fb.07.55.00.b7.8c.00.00.00.00.00.00.19.04
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: transform_listener_impl_5615f85e69e0
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: SUBSCRIPTION
GID: 01.0f.fb.07.75.00.de.2d.00.00.00.00.00.00.18.04
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: rosbridge_websocket
Node namespace: /
Topic type: tf2_msgs/msg/TFMessage
Topic type hash: RIHS01_e369d0f05a23ae52508854b66f6aa0437f3449d652e8cbf22d5abe85d020f087
Endpoint type: SUBSCRIPTION
GID: 01.0f.fb.07.ee.0a.be.cd.00.00.00.00.00.00.42.04
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: TRANSIENT_LOCAL
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

I'm putting the code part below that causes this problem also
(rosbridge_suite/rosbridge_library/src/rosbridge_library/internal/subscribers.py)

```
if any(pub.qos_profile.durability == DurabilityPolicy.TRANSIENT_LOCAL for pub in infos):
    qos.durability = DurabilityPolicy.TRANSIENT_LOCAL
    qos.reliability = ReliabilityPolicy.RELIABLE
if any(pub.qos_profile.reliability == ReliabilityPolicy.BEST_EFFORT for pub in infos):
    qos.reliability = ReliabilityPolicy.BEST_EFFORT
```

As you can see in this part, if there is any publisher with a TRANSIENT_LOCAL durability policy, the code sets the subscriber's durability policy to TRANSIENT_LOCAL and its reliability to RELIABLE.

bjsowa, 2025-03-28:
What's the `velocity_controller` node and why is it publishing to `/tf` topic with TRANSIENT_LOCAL durability policy?

Egoless41, 2025-03-28:
It's not a custom package. `velocity_controller` is a library of `ros2_control` named as `controller_manager`. In `ros2_control`, `tf` publisher initialized with RELIABLE reliability policy and TRANSIENT_LOCAL durability policy.

Egoless41, 2025-04-15:
Solved in #1023.

closed 2025-04-15 by Egoless41