RobotWebTools/rosbridge_suite #1026: rosbridge_websocket-1]: process has died [pid 10904, exit code -11

chentaixian, 2025-04-01:
[ERROR] [rosbridge_websocket-1]: process has died [pid 10904, exit code -11, cmd '/home/unitree/ros2_ws/install/rosbridge_server/lib/rosbridge_server/rosbridge_websocket --ros-args -r __node:=rosbridge_websocket --params-file /tmp/launch_params_9uvka9ji --params-file /tmp/launch_params_nr2nnzaj --params-file /tmp/launch_params_sanz27xi --params-file /tmp/launch_params_vw6tvjcu --params-file /tmp/launch_params_dy95tmr4 --params-file /tmp/launch_params_5mcte0b2 --params-file /tmp/launch_params_uwbu42wf --params-file /tmp/launch_params_qi3ry31n --params-file /tmp/launch_params_b2bh507i --params-file /tmp/launch_params_ys8exyis --params-file /tmp/launch_params_c6t4ak9r --params-file /tmp/launch_params_ieu475sd --params-file /tmp/launch_params_doa40o9g --params-file /tmp/launch_params_6vi3j0wz --params-file /tmp/launch_params_64km3uri'].

chentaixian, 2025-04-01:
1

bjsowa, 2025-04-01:
I'm gonna need a little more info than that...

chentaixian, 2025-04-01:
> I'm gonna need a little more info than that...

The rosbridge service starts normally, but after sending an action goal (send_goal) a few times via ROS actions, the system throws an error. No other error messages or logs are observed beyond what has been described, and I cannot pinpoint the root cause.

bjsowa, 2025-04-01:
Foxy is EoL for almost 2 years now. The last version of rosbridge_suite released to this distro is `1.3.1` so you might try your luck with that instead of the latest commit.

If the issue still persists with the older version, it would help if you could [get a backtrace](https://docs.nav2.org/tutorials/docs/get_backtrace.html)

chentaixian, 2025-04-02:
> Foxy is EoL for almost 2 years now. The last version of rosbridge_suite released to this distro is `1.3.1` so you might try your luck with that instead of the latest commit.
>
> If the issue still [...]

I'm currently using version 1.3.1. However, my ROS2 is the Foxy distribution. I’m not sure if this is causing the error

XW-Ming, 2025-04-09:
> Foxy is EoL for almost 2 years now. The last version of rosbridge_suite released to this distro is `1.3.1` so you might try your luck with that instead of the latest commit.
>
> If the issue still [...]

GDB stacktrace is following：
ming@LAPTOP-7L6NF3CQ:~/ros2_ws$ gdb python3 core-python3-1744198480
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.2) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...
(No debugging symbols found in python3)

warning: core file may not match specified executable file.
[New LWP 34066]
[New LWP 34108]
[New LWP 34104]
[New LWP 34107]
[New LWP 34105]
[New LWP 34106]
[New LWP 34111]
[New LWP 34110]
[New LWP 34172]
[New LWP 34109]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `python3 /home/ming/ros2_ws/install/rosbridge_server/lib/rosbridge_server/rosbri'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f264c52bd7d in rcl_client_is_valid () from /opt/ros/foxy/lib/librcl.so
[Current thread is 1 (Thread 0x7f264cd4f040 (LWP 34066))]
(gdb) bt tree
No symbol "tree" in current context.
(gdb) bt full
#0  0x00007f264c52bd7d in rcl_client_is_valid () from /opt/ros/foxy/lib/librcl.so
No symbol table info available.
#1  0x00007f264c577ea8 in rcl_action_client_is_valid () from /opt/ros/foxy/lib/librcl_action.so
No symbol table info available.
#2  0x00007f264c579f60 in rcl_action_client_wait_set_get_num_entities () from /opt/ros/foxy/lib/librcl_action.so
No symbol table info available.
#3  0x00007f264c58bc51 in ?? () from /opt/ros/foxy/lib/python3.8/site-packages/rclpy/_rclpy_action.cpython-38-x86_64-linux-gnu.so
No symbol table info available.
#4  0x00000000005f596a in PyCFunction_Call ()
No symbol table info available.
#5  0x00000000005f6496 in _PyObject_MakeTpCall ()
No symbol table info available.
#6  0x0000000000570807 in _PyEval_EvalFrameDefault ()
No symbol table info available.
#7  0x00000000005f5c76 in _PyFunction_Vectorcall ()
No symbol table info available.
#8  0x000000000056b356 in _PyEval_EvalFrameDefault ()
No symbol table info available.
#9  0x0000000000500714 in ?? ()
No symbol table info available.
#10 0x00000000004f5033 in ?? ()
No symbol table info available.
#11 0x00000000005c4220 in ?? ()
No symbol table info available.
#12 0x000000000056b175 in _PyEval_EvalFrameDefault ()
No symbol table info available.
[... 81 lines cut ...]
No symbol table info available.
#53 0x000000000068ce77 in PyEval_EvalCode ()
No symbol table info available.
#54 0x000000000067e211 in ?? ()
No symbol table info available.
#55 0x000000000067e28f in ?? ()
No symbol table info available.
#56 0x000000000067e331 in ?? ()
No symbol table info available.
--Type <RET> for more, q to quit, c to continue without paging--
#57 0x000000000067e6d7 in PyRun_SimpleFileExFlags ()
No symbol table info available.
#58 0x00000000006b6d42 in Py_RunMain ()
No symbol table info available.
#59 0x00000000006b70cd in Py_BytesMain ()
No symbol table info available.
#60 0x00007f264cf3d083 in __libc_start_main (main=0x4eead0 <main>, argc=35, argv=0x7ffec8fe1988, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffec8fe1978) at ../csu/libc-start.c:308
        self = <optimized out>
        result = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {7041248, -1006427170756286646, 6269968, 140732270516608, 0, 0, 1005958779982128970, 884088943088969546}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x23, 0x7ffec8fe1988}, data = {prev = 0x0, cleanup = 0x0, 
              canceltype = 35}}}
        not_first_call = <optimized out>
#61 0x00000000005fac3e in _start ()
No symbol table info available.
(gdb)