RobotWebTools/rosbridge_suite #1036: rosbridge and roslibpy do not work with actions

kineticsystem, 2025-05-20:
**Description**

I'm trying to use roslibpy to send a goal to an action server. ROS Bridge seems to be looking for the definition of the action in the wrong place.

- Library Version: latest on branch "humble"
- ROS Version: humble
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**

I created a dummy project to reproduce the problem:
https://github.com/kineticsystem/ros-brigde-poc

The project shows that using rclpy the action server works as expected, but not when using roslibpy.

**Expected Behavior**

The action server receives the Action goal and returns a feedback stream with the final result.

**Actual Behavior**

```bash
[rosbridge_websocket-3] [ERROR] [1747735113.080290111] [rosbridge_websocket]: [Client 14a428ca-51cc-4f4d-ae4d-1122626ff31c] [id: subscribe:/test_action/result:5] subscribe: Unable to import bridge_msgs.msg from package bridge_msgs. Caused by: No module named 'bridge_msgs.msg'
```

It seems a bug to me, unless we are misusing roslibpy. I'm not looking for a solution, but some feedback would be helpful while trying to fix the problem.

bjsowa, 2025-05-20:
ROS 2 actions are not supported by roslibpy yet. (see gramaziokohler/roslibpy#122)

kineticsystem, 2025-05-20:
Thank you, that explains all.

sea-bass, 2025-06-13:
Wait! There is a long standing branch in roslibpy that does support actions

https://github.com/gramaziokohler/roslibpy/pull/125

The maintainers just happen to be super inactive and it hasnt been reviewed/merged.

@kineticsystem you should be able to directly use that branch

EzraBrooks, 2025-07-14:
This is not a limitation of rosbridge itself (Actions work fine with roslibjs!)

closed 2025-07-14 by EzraBrooks