RobotWebTools/rosbridge_suite #1055: [DEVELOPERS ONLY] Who is working on which repository

MatthijsBurgh, 2025-07-28:
The teams in this organization are a bit of mess. I am trying to improve this. To get a better overview, please use this issue to let me know on which repositories you are working.

Please also let me know what level of permissions you think you should have. Or what level other people should (not) have.

I also hope to assign maintainers/admin for each repository.

EzraBrooks, 2025-11-04:
At least for the moment, I'm less busy at my new role than my last one and am investing a bit more time here to clean up the roslibjs backlog - and if I can chew through that I might move onto some tf2_web_republisher and rosbridge stuff.

EzraBrooks, 2025-11-04:
Ah, whoops, I meant ros3djs - I typed tf2_web_republisher because the most recent issue I saw regarding ros3djs was incompatibility with the ROS 2 version of the TF republisher :)

ceoifung, 2025-11-17:
tf2_web_republisher has big program in ros jazzy version. when subscribe tfclient, error occur: 
```shell
[rosbridge_websocket-8] [ERROR] [1763351602.510417846] [rosbridge_websocket]: [Client fd19f02a-481e-4aa8-ae8d-9a589736e80f] [id: advertise:/tf2_web_republisher/goal:1] advertise: Unable to import tf2_web_republisher.msg from package tf2_web_republisher. Caused by: No module named 'tf2_web_republisher'
[rosbridge_websocket-8] [ERROR] [1763351602.531161138] [rosbridge_websocket]: [Client fd19f02a-481e-4aa8-ae8d-9a589736e80f] [id: subscribe:/tf2_web_republisher/feedback:3] subscribe: Unable to import tf2_web_republisher.msg from package tf2_web_republisher. Caused by: No module named 'tf2_web_republisher'
[hclidar_driver_ros2_node-11] start_time=1763351602364407587, end_time=1763351602554385525 , diff_offset=0
[hclidar_driver_ros2_node-11] Main: total Rx Points=403 ,
[cartographer_node-13] [INFO] [1763351602.584979524] [cartographer logger]: I20251117 11:53:22.-2147483648 47151 collated_trajectory_builder.cc:81] odom rate: 48.56 Hz 2.06e-02 s +/- 9.72e-03 s (pulsed at 100.17% real time)
[cartographer_node-13] [INFO] [1763351602.585044247] [cartographer logger]: I20251117 11:53:22.-2147483648 47151 collated_trajectory_builder.cc:81] scan rate: 5.18 Hz 1.93e-01 s +/- 4.75e-03 s (pulsed at 99.93% real time)
[rosbridge_websocket-8] [INFO] [1763351602.661145752] [rosbridge_websocket]: [Client fd19f02a-481e-4aa8-ae8d-9a589736e80f] Subscribed to /scan
[hclidar_driver_ros2_node-11] start_time=1763351602554385525, end_time=1763351602754408800 , diff_offset=0
[hclidar_driver_ros2_node-11] Main: total Rx Points=403 ,
[rosbridge_websocket-8] [INFO] [1763351602.814612347] [rosbridge_websocket]: [Client fd19f02a-481e-4aa8-ae8d-9a589736e80f] Subscribed to /map
[rosbridge_websocket-8] [ERROR] [1763351602.816397028] [rosbridge_websocket]: [Client fd19f02a-481e-4aa8-ae8d-9a589736e80f] [id: publish:/tf2_web_republisher/goal:6] publish: Cannot infer topic type for topic /tf2_web_republisher/goal as it is not yet advertised
[rosbridge_websocket-8] [ERROR] [1763351602.818141782] [rosbridge_websocket]: [Client fd19f02a-481e-4aa8-ae8d-9a589736e80f] [id: publish:/tf2_web_republisher/goal:8] publish: Cannot infer topic type for topic /tf2_web_republisher/goal as it is not yet advertised
```
I check tf2_web_republisher source code. found that msg was move to tf2_web_republisher_interface, so cannot find  tf2_web_republisher.msg. I try to fix that. Move the tf2_web_republisher_interface/msg, action srv to tf2_web_republisher like the repo https://github.com/Wisc-HCI/tf2_web_republisher_py do. It build succeed but another error occur

```shell
[Client 14703ea6-658f-4e97-b678-f0be50b8d50a] [id: advertise:/tf2_web_republisher/goal:1] advertise: Unable to import msg class TFSubscriptionActionGoal from package tf2_web_republisher. Caused by module 'tf2_web_republisher.msg' has no attribute 'TFSubscriptionActionGoal'
[rosbridge_websocket-4] [ERROR] [1763365964.477028133] [rosbridge_websocket]: [Client 14703ea6-658f-4e97-b678-f0be50b8d50a] [id: subscribe:/tf2_web_republisher/feedback:3] subscribe: Unable to import msg class TFSubscriptionActionFeedback from package tf2_web_republisher. Caused by module 'tf2_web_republisher.msg' has no attribute 'TFSubscriptionActionFeedback'
[rosbridge_websocket-4] [ERROR] [1763365964.561615087] [rosbridge_websocket]: [Client 14703ea6-658f-4e97-b678-f0be50b8d50a] [id: publish:/tf2_web_republisher/goal:6] publish: Cannot infer topic type for topic /tf2_web_republisher/goal as it is not yet advertised
[rosbridge_websocket-4] [ERROR] [1763365964.852200166] [rosbridge_websocket]: [Client 14703ea6-658f-4e97-b678-f0be50b8d50a] [id: publish:/tf2_web_republisher/goal:8] publish: Cannot infer topic type for topic /tf2_web_republisher/goal as it is not yet advertised
```

In ros2 jazzy version, when generate action files, TFSubscriptionActionGoal change to TFSubscription_Goal. but the rosbridge_suite is incompatibility. When i use TFSubscription_Goal to fix this error. report that
```shell
 [Client 2264b1b2-8136-4b65-8020-5b2e03fd3d5a] [id: publish:                                                               /tf2_web_republisher/goal:6] publish: Message type tf2_web_republisher/TFSubscription_Goal does not have a field goal_id
```
Any solutions? are my method to solve this program is wrong?

MatthijsBurgh, 2026-02-02:
@chfritz I will do this in 1-2 days.

minggangw, 2026-02-03:
Hi @MatthijsBurgh Sorry for the delay, I mainly maintian [rclnodejs ](https://github.com/RobotWebTools/rclnodejs)  and [rclnodejs-cli](https://github.com/RobotWebTools/rclnodejs-cli) projects, if anything needs me, please ping me here. Another repo, [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge), was also initiated by me, but it's out of maintenance now, thanks!